bsc-networks/mesh #8: Configure static IP on the Wilde-Loth antenna

mitar, 2015-04-05:
I propose `10.20.15.210`.

Document it here: https://github.com/bsc-networks/mesh/blob/master/README.md

See Cloyne-Kingman configuration:

[image]

CdavM, 2015-04-06:
Assigned it to 10.20.15.210
[image]

mitar, 2015-04-11:
We moved the IP pool. We have to renumber again.

CdavM, 2015-04-13:
Hm. I can't connect to the antenna anymore. I was able to connect to it when I set it up, but I don't know what happened in the meantime. I tried connecting directly to it (setting my IP to 10.20.15.209; gateway, dns and subnet the same as in the antenna settings), but I still couldn't reach it. Any suggestions?

mitar, 2015-04-13:
Is it on? :-)

mitar, 2015-04-13:
Strange. But good that we have it back.