bsc-networks/mesh #10: Expand the IP pool for all bsc houses

CdavM, 2015-04-11:
Let's plan ahead and create IP slots for all BSC units. Apartments usually don't have a single internet connection, right? Should we allocate a /22 mask for everyone?

CdavM, 2015-04-11:
I added the rest of the houses and apartments. I sorted them alphabetically. Do we want to reassign the current IP's so we'll have a nice sorted list?

mitar, 2015-04-11:
I think `/22` for everyone is good. This is how I envisioned it. It does not matter if apartments do not have a single Internet connection, they can still share one IP pool.

> I added the rest of the houses and apartments. I sorted them alphabetically. Do we want to reassign the current IP's so we'll have a nice sorted list?

I do not think we should worry about the order of the list by name. In some way I wanted it to represent the order how people joined. :-)

mitar, 2015-04-11:
Hm, now I am confused why I choose 22 and not 21? 21 gives you nice 32 pools for 32 possible locations which is good for future expansion but also gives a lot to each location. Or do you think 22 is enough? We can also have 22 as a base unit and then allocate to locations which need more two units.

mitar, 2015-04-11:
But then we might want to allocate each house with an adjacent empty IP pool. To allow future expansion.

mitar, 2015-04-11:
Maybe something like:

* `10.20.0/22` – reserved
* `10.20.8/22` – Cloyne
  * `10.20.11.210` – Cloyne-Kingman
* `10.20.16/22` – Kingman
  * `10.20.23.210` – Kingman-Cloyne
* `10.20.24/22` – Wilde
  * `10.20.31.210` – Wilde-Lothlorien
* `10.20.32/22` – Lothlorien
  * `10.20.39.210` – Lothlorien-Wilde
* `10.20.40/22` – Afro
* `10.20.48/22` – Casa Zimbabwe
* `10.20.56/22` – Castro
* `10.20.64/22` – Convent
* `10.20.72/22` – Davis
* `10.20.80/22` – Euclid
* `10.20.88/22` – Fenwick
* `10.20.96/22` – HiP
* `10.20.104/22` – Hoyt
* `10.20.112/22` – Kidd
* `10.20.120/22` – Northside
* `10.20.128/22` – Ridge
* `10.20.136/22` – Rochdale
* `10.20.144/22` – Sherman
* `10.20.154/22` – Stebbins
* `10.20.162/22` – Wolf

CdavM, 2015-04-11:
So you propose to keep everyone on /22 for now?

mitar, 2015-04-11:
(I am lazy moving Cloyne to something new again, because it took me the whole day last weekend.)

mitar, 2015-04-11:
So my proposal here does move Kingman and Wilde and Lothlorien networks. This might not be fair that I am being lazy. So yes, keeping 22, but allowing future expansion into 21 if needed.

mitar, 2015-04-11:
Maybe we should just do it in order, with 22, and one empty pool in between, so that everyone has to move. :-)

CdavM, 2015-04-11:
if we're moving again, should we not do it alphabetically :D ?

mitar, 2015-04-11:
Yes, that I am saying:
* or we do not do anything and just add 22 to other houses (and we have enough room later on for bigger house to move to if they need more)
* or we leave Cloyne where it is and move other houses so that there is 22 space in-between
* or we move everyone, then we can make also alphabetically

The issue is, I am lazy. :-)

CdavM, 2015-04-11:
OK, my suggestion is to go with the plan number 2: Move everyone but Cloyne.

mitar, 2015-04-11:
Ah, this looks so special then for Cloyne. Not really cooperative. Let's move all. :-)

mitar, 2015-04-11:
Please check: https://github.com/bsc-networks/mesh/blob/master/README.md

closed 2015-04-20 by mitar