bsc-networks/mesh #9: Buy equipment

mitar, 2015-04-11:
So I suggest we buy equipment for the rest of the links in bulk. Until now we have been using [PowerBeam PBE-M5-300](http://dl.ubnt.com/datasheets/powerbeam/PowerBeam_DS.pdf) with nominal bitrate of 150 Mbit/s full-duplex. As it has only 100 Mbit/s LAN port the real throughput is 100 Mbit/s full-duplex anyway. Because we have observed that on our distances links work with full speed even if we configure lowest transmission power, we could try to use instead a new device. [NanoBeam NBE-5AC-19](http://dl.ubnt.com/datasheets/NanoBeam_ac/NanoBeam_ac_DS.pdf). It has smaller gain (19 dbi vs. 22 dbi) but it does not matter for our short distances (even aligning would be easier). But it has 450 Mbit/s nominal full-duplex transmission and gigabit port. Because our SNR has enough room for new modulation this will probably be achieved. The question is mostly do we want to use 80 MHz channel width (to really get this bitrate). It just uses a lot of airtime for our neighbors. But the difference in price is only $20 ($80 vs. $100) so we should probably go with it. We can always run it on smaller channel width.

We would also need simple APs/switches/routers which can then be gigabit and that we can run OpenWrt on them to host a routing protocol. (NBE-5AC-19 is not yet supported by the OpenWrt and will probably not be for quite some time.)

mitar, 2015-04-11:
Here is [FCC specification](https://apps.fcc.gov/oetcf/eas/reports/ViewExhibitReport.cfm?mode=Exhibits&RequestTimeout=500&calledFromFrame=N&application_id=753856&fcc_id=SWX-NBEM5AC) for the device. It seems it uses Qualcomm chipset, not Atheros. But this might be just rebranded chipset?

mitar, 2015-04-11:
OK, wireless chipset is QCA9882-BR4A. It seems there is an [ath10k](https://wireless.wiki.kernel.org/en/users/Drivers/ath10k) for it. But I wonder how stable this is?

So one downside is that at least for some time we will not be able to install OpenWrt on them and we will really have to use additional devices to run the routing protocol.

mitar, 2015-04-11:
Yes, it is confirmed. Currently ath10k is still using a binary blob driver and is just a wrapper around it. We do not yet have an open source implementation.

mitar, 2015-04-12:
So the issue is that with going to gigabit, then we also have to buy more expensive routers.

If we assume that we have $1500, then this means:
* $100 + $70 -> 8 sides of links, 4 links
* $80 + $20 -> 15 sides of links, 7 links

We do have gigabit routers at Cloyne though, so we could reuse them, but still. So the question is if we should invest into more links now and then once we fill that we can always replace with better and faster things. And we will also have more choice then about the equipment. I think for now it seems a bit too costly to go with gigabit equipment.

mitar, 2015-04-12:
OK, we have 5 gigabit routers at Cloyne, but I am not sure if I can just use it like this. But with them, we could probably do 11 sides of links with gigabit equipment.

mitar, 2015-04-13:
OK, we decided to go with $100. Let's then do $100 and TL-WR1043ND.

CdavM, 2015-04-13:
OK, let us know when the equipment arrives. Can we paint these dishes?

mitar, 2015-04-13:
We could not put the covers on, then they look almost the same as current ones. I am not sure if we can paint over the cover. We will see when we get them.

mitar, 2015-04-13:
So we have one extra antenna which is currently at Kingman to connect to CZ.

CdavM, 2015-04-13:
A m5-300?

mitar, 2015-04-13:
Yes.

mitar, 2015-04-13:
@jmrtns, what is that other store you found which you sad is cheaper?

CdavM, 2015-04-13:
http://www.amazon.com/s/ref=nb_sb_noss_1?url=search-alias%3Delectronics&field-keywords=Nanobeam+ac

Maybe something like this? :)

mitar, 2015-04-13:
I will order one M5-300 to match the one we already have.

CdavM, 2015-04-13:
Fine, but let's put the slow link to the 'leaf node' of our tree.

CdavM, 2015-04-19:
Hi! We should order the equipment, I'd like to put it up during dead week :)

mitar, 2015-04-20:
Ordered.

mitar, 2015-04-27:
We got 7 TP-Links and 1 NanoBeam AC. The latter are really small! Like my palm size. :-) They really look cool. I am not sure about painting, but we will see.

The rest should come soon.

I also have the PBE-M5-300 from Kingman.

mitar, 2015-04-27:
I got a confirmation that we can paint and that we should not worry about it. :-) It does not really matter which type of color, but acrylic paint is the best.

mitar, 2015-04-27:
We got 7 more NanoBeam ACs. So we have 8 now. We have all of them which we ordered. We are still waiting for the mounts and the PBE-M5-300.

mitar, 2015-04-30:
We got also the PBE-M5-300. Now only the mounts. Probably it is not critical to have them with this latest equipment.

We gave:
* one NanoBeam AC to Loth
* one NanoBeam AC to Sherman
* two NanoBeam AC to Castro
* two NanoBeam AC to Afro

CdavM, 2015-05-03:
OK, we should also exchange the cable for a better one. Here's the one we used yesterday:

http://www.amazon.com/Monoprice-1000-Feet-500Mhz-Copper-Ethernet/dp/B008I8AJIY/

jmrtns, 2016-10-04:
* It seems that OpenWRT is no longer on our list of prerequisites since tp-link or mikrotik routers will handle everything OpenWRT on the access points would have and OpenWRT isn't the best on many of these antennas. So we seem to just be sticking to NanoBeam ACs

  * We should probably buy some proper exterior CAT6 for these antennas and then patch into cheaper interior CAT6a or just get exterior CAT6a. primuscable sells exterior cable but exterior cable doesn't seem to use standard heads to well since the individual wires are thicker. primus sells exterior heads but they are about $0.25 each.

mitar, 2016-10-05:
So, for antennas, yes, OpenWRT does not work so well on them, so we are going with design where we have antennas in bridge mode, and then a router with OpenWRT behind them.