bsc-networks/mesh #12: Set up Loth - Afro, Afro - Castro, Castro - Sherman, Sherman - Davis links

CdavM, 2015-04-27:
The planned date is Saturday, May 02, at 10am. We meet at Wilde. We'll see how far we get.

mitar, 2015-04-27:
Should we try to organize that network cables are put in place in advance?

The antennas are pretty small, so probably we can reuse some of existing poles.

How about renumbering of networks.

We should start inviting people to paint the antennas. I can ask around Cloyne. But we should probably first ask at each respective house.

CdavM, 2015-04-28:
Organizing network cables ahead of time might be too ambitious, especially
since we don't know where the antennas are going to go. But we can do two
things (mount antenna; lay cables) in parallel, so there's no bottlenecks
here.

I've sent an email asking them to get someone to paint them and to renumber
their antennas. I'm pessimistic about their response, so it would be good
to start asking around at Cloyne. We have a lot to paint anyway :)

CdavM, 2015-05-03:
Current status of the links:
* [x] Loth - Afro
* [x] Afro - Castro
* [x] Castro - Sherman
* [ ] Sherman - Davis (network cable)

mitar, 2015-05-03:
I just updated the checkboxes in the description. Then GitHub also nicely shows the progress in the ticket list.

CdavM, 2015-05-03:
@nherson 
Regarding the cable: Wilde has 140ft of CAT6 cable you can use. If that suffices, let me know. If not, @mitar is ordering more soon.

nherson, 2015-05-04:
If I remember correctly, I have a massive roll of CAT6 laying around, but I don't have heads for it.

nherson, 2015-05-04:
Also, @mitar mentioned an old and dead cable link between Davis and Sherman.  Where was that? I had no idea.

mitar, 2015-05-04:
I have heads/connectors for CAT6.

It is just going there from the third floor Sherman to second floor Davis. I am not sure how to describe it.

nherson, 2015-05-04:
Ok.  I am going to try to run a cable from the Davis attic across the alley way.  I can work on it during dead week. Any chance you can give me the MAC address of the router I'll be getting so that I can assign it an IP?

mitar, 2015-05-04:
Routers will have static IPs configured, so that they do not depend on house network being up for them to work. Your house mesh router will have IP `10.20.51.200`.

If you want, I can come and help you during the dead week. Probably few others can also join.