bsc-networks/mesh #15: Power all devices down

mitar, 2015-05-03:
It seems all devices have enough SNR that we could power them all the way down.

CdavM, 2015-05-03:
I lowered the power to -4 and set the channel width to 10MHz for all antennas. The current signal is ~ -50dbi, noise ~ -90dbi. Airmax quality is still 100%, but airmax capacity is now at 90%. I suggest we keep it at 10MHz unless it proves unsatisfactory.

closed 2015-05-03 by CdavM

mitar, 2015-05-03:
Oh, 10 MHz is really low. We used 20 MHz on existing links. So -4 power and 20 MHz link. This should give nice 100 Mbit throughput. So when I look at stations on Cloyne node, with 20 Mhz channels and highest MCS (automatically selected) I get 144.44/144.44 Mbps TX/RX. How much you get for AC with 10 MHz? And how much 20 with Mhz?

And yes, that should be good enough for now and then we can start changing things.

CdavM, 2015-05-03:
All TX/RX rates are listed as: 8x (256QAM MIMO)
Airmax capacity, airmax quality, tx rate and rx rate do not change if we go from 10MHz to 20. (just tested)

mitar, 2015-05-04:
Eh, they do not provide bitrate directly anymore. So we have to compute this manually, [using this spreadsheet](https://en.wikipedia.org/wiki/IEEE_802.11ac#Data_rates_and_speed). So for old antennas, we have 20 Mhz channel, 64-QAM, and two streams, so this gives 72.2  * 2 = 144.

For 256-QAM sadly standard does not even provide 20 Mhz and 10 Mhz channels for 5/6 coding rate, it seems (or is even not possible?). Anyway, the difference between 64-QAM and 256-QAM is not so big that halving the channel size would not be noticeable. So I suggest we still use 20 MHz even with 256-QAM.

CdavM, 2015-05-04:
But old antennas only have 2 spatial streams and the new ones have 8. So assuming the relationship is linear, we should have a theoretical limit of 180MBit (for 40 MHz;800 ns GI) / 4 (to get to 10MHz) * 8 (number of spatial streams) ~ 360Mbit. Let's first see if this is achievable. It's not difficult to change the settings if we reach these speeds.

mitar, 2015-05-04:
New have 8? I think they have only 2 as well. Horizontal and vertical polarization. That's it. No?

The 8x in front of 245QAM does not mean 8 spatial streams, I think. See the discussion here:

https://community.ubnt.com/t5/airMAX-AC/W00T-Just-got-a-pair-of-Nanobeam-M5-AC-s-or-Powerbeam/td-p/1031139

> 64QAM or 6x means that for every frame sent it encodes 6 bits of data.   It's roughly equivelent to MCS 5/6/7 (depending on some other things like FEC mode)   16QAM would be like MCS 3/4, and 256QAM is like MCS8/9.   That's all for a single spatial stream - roughly double the MCS numbers for 2 streams.

So yea, it seems old antennas use 6 bits, new 8 bits, so you get some more throughput, but not really twice the amount which would allow you to split the channel width.

mitar, 2015-05-04:
See the screenshot here: https://cloud.githubusercontent.com/assets/5568048/7446332/90065558-f189-11e4-9756-85fc4e3f4fd7.png

You have 2x2 chains, that is 2 streams. The same as old antennas have:

```
TX/RX Chains: 2X2
```

The main reason why AC is promising more throughput is because of the 256QAM modulation (8 bits instead of 6) and 80 MHz channels. If we keep all channels to 20 Mhz, there is not much difference between old and new devices. What new devices allow us is to go to wider channels when we will need that.

mitar, 2015-05-04:
BTW, all above is what I am understanding by reading around the links I am posting to you. I am not sure of anything of this. So if you find some better explanations, please share. But from what I understand, we have only two spatial streams, two chains (vertical and horizontal polarization), and we then can use various channel width and various modulations. Modulations can give you a bit more throughput, but the main influence is still the number of streams and channel width.

Of course, signal-to-noise ratio is the one which tells you which modulation you can use. So more dense modulation more signal-to-noise you need. You can get more by increasing the transmission power. We lowered it all down now, so if we can still achieve 256QAM, that is great!

mitar, 2015-05-04:
Wider the channel, higher is the chance that you get more noise, so lower signal-to-noise.

CdavM, 2015-05-04:
Ah, okay, this makes sense. I'd love to do some tests for actual throughput and compare the different modes.

mitar, 2015-05-04:
But yes, I am not sure why they display `8x` when they mean 8 bits. :-) And I am not sure why they display `2x2` when they mean two streams/chains and `2x1` when they mean one stream/chain.

mitar, 2015-05-04:
cc @musti? :-)

mitar, 2015-05-04:
And yes, we should measure things. Probably we should move all antennas to the `/16` subnet. But let's wait a bit to get feedback from @kostko in #14.