bsc-networks/mesh #20: Link between CZ and Hoyt

mitar, 2015-06-22:
I talked to people at Hoyt and they are interested in making a link to CZ. There is very short distance to Euclid, so we could also throw a network cable there, and the backyard neighbors to northside apartments.

jmrtns, 2016-10-04:
So there would be a link from Hoyt to both CZ and Euclid correct? I haven't been able to contact HOYNM yet in regards to this.

mitar, 2016-10-05:
I think I would for now go with a cable between Hoyt and Euclid. Can you go and try to reach them? Also talk to Stebbins and let's connect them as well. I think they are more active this semester.

jmrtns, 2016-10-09:
When I said I haven't been able to contact HOYNM, I meant that I'de tried multiple times (twice digitally and once in person) to contact them but they have yet to respond. Stebbins is to connect to both to Cloyne and Kidd correct? Are we bringing northside coops into this? I believe Northside coops could connect to Euclid, Hoyte, Kidd, and possibly Stebbins also.

mitar, 2016-10-09:
Yes, we could connect Stebbins to Cloyne, I think. So whatever is reasonable.

We should talk to Northside coops as well.

Can you take this over and get into contact with them?

jmrtns, 2016-10-10:
Who is the network contact of Northside coops? Any suggestions on how to contact Hoynm?

mitar, 2016-10-13:
I do not know. I know that @CdavM was helping a bit on northside. Ask him.

CdavM, 2016-10-13:
Hi!

They don't have any network people, so the best contact is their house manager. She's quite nice, but she probably won't have time to help out with cables / stuff.