bsc-networks/mesh #21: Field Trip to Internet Archive, 2/12

CdavM, 2016-02-05:
Hi everyone!

I'm going to the Internet Archive next Friday to have lunch with them. Lunch starts at noon. Everyone's invited to join me on the adventure! Oh, if anyone can arrange a ride there I'd be very grateful.

CdavM, 2016-02-05:
@mitar: maybe @sunnysideofthescreen or @jmrtns want to come? can you poke them :)

jmrtns, 2016-02-05:
I has car and want to go

mitar, 2016-02-05:
I would also like to invite Graham. Should we also send it to bsc-hackers and hackers@cloyne mailing lists?

mitar, 2016-02-05:
Can you prepare a message I would send?

CdavM, 2016-02-05:
yes invite everyone!!

message subject: Field trip to Internet Archive on 2/12

Hi everyone,

I'm planning to go to the Internet Archive (SF) next Friday for lunch. It's a non-profit that is trying to archive human knowledge in all its forms. If anyone is interested in coming with me, let me know and get your inner geek out! It could be fun to see where a backup copy of the Internet is stored. We have to be there by noon.

mitar, 2016-02-05:
Sent.

I do not have time tomorrow.

CdavM, 2016-02-05:
That's ok because this is happening next week :)

mitar, 2016-02-05:
Oh. :-) Good. :-)

closed 2017-05-12 by mitar