bsc-networks/mesh #23: Updaye README with euclid antenas

jmrtns, 2016-10-04:
I need to change the IP of the Euclid-Cloyne station to 10.20.59.210 to follow the standard a bit more and then I need to find then add the MACs for the Euclid CZ link. This will have to be after I have access to the antennas which I currently do not.

mitar, 2016-10-05:
Why you do not have access to antennas? They should be up on your side, they are just not connected to the switch.

mitar, 2016-10-05:
Hm, maybe this is just a typo in documentation? Are you sure it is not already configured?

And yes, please add Euclid-CZ.

jmrtns, 2016-10-09:
I currently can't access our CZ antenna because some services depend on it and it's beyond our firewall. I was hoping to access everything in one go after the mesh routers are configured instead of plugging cables around and messing with VLANs

clonm, 2020-03-18:
This has been updated

closed 2020-03-18 by clonm